lorenwel/linefit_ground_segmentation #2: should there be a ()

rambleramble, 2017-12-29:
linefit_ground_segmentation/src/segment.cc   [line47](https://github.com/lorenwel/linefit_ground_segmentation/blob/11e4ed0090f7637a5929b3c56febe9a6687941ab/linefit_ground_segmentation/src/segment.cc#L47)

Wondering if it should be a bracket wrapping around the last two bools such as following 
`if (error > max_error_ ||
            std::fabs(cur_line.first) > max_slope_ ||
            (is_long_line && std::fabs(expected_z - cur_point.z) > max_long_height_)     ) `

lorenwel, 2017-12-29:
The logical AND operator has precedence over the logical OR. The expression with brackets is equivalent to the one without. http://en.cppreference.com/w/cpp/language/operator_precedence

closed 2017-12-29 by lorenwel

rambleramble, 2017-12-29:
Yep, thanks a lot for the prompt reply.