lorenwel/linefit_ground_segmentation #3: segment.cc > no good line condition

rambleramble, 2017-12-29:
This condition below covers not good line condition.
[https://github.com/lorenwel/linefit_ground_segmentation/blob/master/linefit_ground_segmentation/src/segment.cc#L47](url)

Wondering why following two segments need t be included in this condition (segment.cc > ln50-ln59) ?
`// Don't let lines with 2 base points through.
          if (current_line_points.size() >= 3) {
            const LocalLine new_line = fitLocalLine(current_line_points);
            lines_.push_back(localLineToLine(new_line, current_line_points));
            cur_ground_height = new_line.first * current_line_points.back().d + new_line.second;
          }`

`// Start new line.
          is_long_line = false;
          current_line_points.erase(current_line_points.begin(), --current_line_points.end());
          --line_iter;`

I thought the above two segments should be executed for good line. Correct me if wrong. Thank you very much in advance!

lorenwel, 2018-01-03:
Those lines of code are executed when we previously had a valid line but adding a new point would make the line invalid (meaning it would violate the ground constraints).
In this case we still want to add the line we had pior to considering the last point as a valid ground line. 

Essentially, we keep adding points to the line until the condition is violated, at which point we save the line assembled up to that point.

rambleramble, 2018-01-03:
Makes sense, thanks so much!

closed 2018-01-03 by lorenwel