lorenwel/linefit_ground_segmentation #5: Bin/segment indices out of bound

YoshuaNava, 2018-05-08:
Hi,
First of all, thank you for sharing your algorithm! I'm currently using it for interfacing ethz-asl/SegMap with a custom LiDAR SLAM pipeline that I developed for my thesis. It has worked great.

When I first tried to use your pkg for segmenting out Velodyne pointclouds, as those from KITTI, I got some issues with the bin and segment indices going out of bound:
https://github.com/lorenwel/linefit_ground_segmentation/blob/11e4ed0090f7637a5929b3c56febe9a6687941ab/linefit_ground_segmentation/src/ground_segmentation.cc#L268

I have submitted a pull request with a patch for this. #4

Kind regards,
Yoshua Nava

rdube, 2018-05-08:
@YoshuaNava thank you for the contribution! It could be nice if you could share some results using this ground removal technique on the segmap repo! Cheers!

YoshuaNava, 2018-05-08:
Hi @rdube 
Thank you!

For sure. I'm submitting the changes I made to the package today. Later today I'll share a video showing the segmentation results I'm getting.

rdube, 2018-05-08:
@YoshuaNava sounds great!

lorenwel, 2018-05-08:
Thanks for your bug hunting. 
Do you have information if bin or segment index or both where out of bounds?

Looking at the code, this should happen when a point has `x<0` and `y==0`. Because the std::atan2 function returns values in the range ]-pi, pi] this should lead to an out of bounds index. The segment index should be 0 though. 

Could you test if adding this check instead of yours fixes this issue? This would avoid us having to discard points.

YoshuaNava, 2018-05-29:
Hi @lorenwel,
I haven't been able to test your suggestion yet. I'm writing my master thesis report now. As soon as I finish I'll do the test.

lorenwel, 2019-02-15:
#11 should have fixed this

closed 2019-02-15 by lorenwel