lorenwel/linefit_ground_segmentation #10: ground_segmentation_test_node.cc:9: undefined reference to `google::InitGoogleLogging(char const*)'?

chowkamlee81, 2018-11-29:
When i build caode using catkin_make got the error below. KIndly help

In function `main':
/home/administrator/catkin_ws1/src/linefit_ground_segmentation_ros/src/ground_segmentation_test_node.cc:9: undefined reference to `google::InitGoogleLogging(char const*)'
/home/administrator/catkin_ws1/src/linefit_ground_segmentation_ros/src/ground_segmentation_test_node.cc:43: undefined reference to `GroundSegmentation::GroundSegmentation(GroundSegmentationParams const&)'
/home/administrator/catkin_ws1/src/linefit_ground_segmentation_ros/src/ground_segmentation_test_node.cc:46: undefined reference to `GroundSegmentation::segment(pcl::PointCloud<pcl::PointXYZ> const&, std::vector<int, std::allocator<int> >*)'
collect2: error: ld returned 1 exit status
linefit_ground_segmentation_ros/CMakeFiles/ground_segmentation_test_node.dir/build.make:708: recipe for target '/home/administrator/catkin_ws1/devel/lib/linefit_ground_segmentation_ros/ground_segmentation_test_node' failed
make[2]: *** [/home/administrator/catkin_ws1/devel/lib/linefit_ground_segmentation_ros/ground_segmentation_test_node] Error 1
CMakeFiles/Makefile2:377: recipe for target 'linefit_ground_segmentation_ros/CMakeFiles/ground_segmentation_test_node.dir/all' failed
make[1]: *** [linefit_ground_segmentation_ros/CMakeFiles/ground_segmentation_test_node.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
CMakeFiles/ground_segmentation_node.dir/src/ground_segmentation_node.cc.o: In function `main':
/home/administrator/catkin_ws1/src/linefit_ground_segmentation_ros/src/ground_segmentation_node.cc:41: undefined reference to `google::InitGoogleLogging(char const*)'
CMakeFiles/ground_segmentation_node.dir/src/ground_segmentation_node.cc.o: In function `SegmentationNode::scanCallback(pcl::PointCloud<pcl::PointXYZ> const&)':
/home/administrator/catkin_ws1/src/linefit_ground_segmentation_ros/src/ground_segmentation_node.cc:23: undefined reference to `GroundSegmentation::GroundSegmentation(GroundSegmentationParams const&)'
/home/administrator/catkin_ws1/src/linefit_ground_segmentation_ros/src/ground_segmentation_node.cc:26: undefined reference to `GroundSegmentation::segment(pcl::PointCloud<pcl::PointXYZ> const&, std::vector<int, std::allocator<int> >*)'
collect2: error: ld returned 1 exit status
linefit_ground_segmentation_ros/CMakeFiles/ground_segmentation_node.dir/build.make:708: recipe for target '/home/administrator/catkin_ws1/devel/lib/linefit_ground_segmentation_ros/ground_segmentation_node' failed
make[2]: *** [/home/administrator/catkin_ws1/devel/lib/linefit_ground_segmentation_ros/ground_segmentation_node] Error 1
CMakeFiles/Makefile2:414: recipe for target 'linefit_ground_segmentation_ros/CMakeFiles/ground_segmentation_node.dir/all' failed
make[1]: *** [linefit_ground_segmentation_ros/CMakeFiles/ground_segmentation_node.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2

lorenwel, 2018-11-29:
Hi, 

we recommend using catkin-tools to build this package. You can install it with `sudo apt install python-catkin-tools`. When installed you can build by calling `catkin build` from the same directory you would call catkin_make. You should clean your workspace before building this way if you have previously built with catkin_make since the two build methods are not compatible.

Make sure you have the glog_catkin package in you workspace and built. 
I don't quite remember but it might be necessary for you to also install glog manually (`sudo apt install libgoogle-glog-dev`).

lorenwel, 2019-02-15:
Closed due to issue being stale

closed 2019-02-15 by lorenwel