lorenwel/linefit_ground_segmentation #15: Does it thread safe for assign segments_[segment_index][bin_index]?

tornadomeet, 2019-05-20:
hello,  in this line : 
https://github.com/lorenwel/linefit_ground_segmentation/blob/master/linefit_ground_segmentation/src/ground_segmentation.cc#L268

It may bring incorrect when multi-thread assign the same ```segment_index``` and ```bin_index```, so does this have an effect on the results? @lorenwel  thanks.

lorenwel, 2019-06-19:
Thanks for noticing. Indeed, the function in question is not thread safe. 
I am swamped at the moment and will revisit this issue in about a month. 

Feel free to open a PR if you fix/have fixed this issue int he meantime.

lorenwel, 2019-10-28:
Hi @tornadomeet , 
I revisited this issue now and noticed that the current implementation is, in fact, thread safe. 
All members of the `Bin` class are std::atomic<> and as such, operations on should be thread safe. 

Feel free to re-open the issue in case I missed somethign.

closed 2019-10-28 by lorenwel