lorenwel/linefit_ground_segmentation #17: Error launching line_fit_ros

BrunoEduardoCSantos, 2019-05-23:
Hi,
I launched line_fit_ros in a separate workspace in ROS and I obtained the following error:
```

[segmentation.launch] is neither a launch file in package [linefit_ground_segmentation_ros] nor is [linefit_ground_segmentation_ros] a launch file name
The traceback for the exception was written to the log file
```

What could be the issue?
Regards,
Bruno

closed 2019-05-23 by BrunoEduardoCSantos

chowkamlee81, 2020-06-05:
I stuck with the same proble,. @brunoeducsantos  kindly help me

lorenwel, 2020-06-05:
Hi, 

this very much looks like a ROS issue and not a problem with this repo.
Make sure you source your workspace after building the package.
`source ~/.bashrc` or `source CATKIN_WS/devel/setup.bash`.
If this doesn't fix it I think you're better of asking on answers.ros.org