lorenwel/linefit_ground_segmentation #18: Ouster cloud segmentation

BrunoEduardoCSantos, 2019-05-29:
Hi @lorenwel ,
Thanks for the amazing work.
I have been using your ros package with ouster LiDAR and there is a typical performance of 100ms. 
For reducing this time, I used threads in addition to the ones you used.
Is there any other way to improve performance of the implementation ?
Regards,
Bruno

lorenwel, 2019-06-03:
Hi Bruno, 
happy to hear you like the code. 

It's a bit hard to judge the runtime without knowing more about your target platform, but that computation time does seem suspiciously high. 
It was a while since I benchmarked my code, but at the time I averaged 8.3ms on an i7 Desktop machine with maybe half the points that an Ouster generates. 
Did you make sure you built in Release mode and that visualization is disabled?

The only other parameters that could influence performance are the segment and bins counts. But those also affect segmentation performance, so you'll have to try what works for you.

If none of this improves the speed significantly, it would be very helpful if you could profile the code as described [here](https://github.com/ethz-asl/programming_guidelines/wiki/Profiling-Code), and post the results.

BrunoEduardoCSantos, 2019-06-07:
Thanks for your feedback. The actually time performance was around 8 ms.
I will close then this issue

closed 2019-06-07 by BrunoEduardoCSantos