lorenwel/linefit_ground_segmentation #21: Segmentation Fault

seboz123, 2019-11-11:
I tried to run the node, but it seems to crash. It processes like 5-8 clouds and then crashes. When using gdb, it throws and segmentation fault. Do you know a solution?

lorenwel, 2019-11-11:
I need more information to be able to help you. 

Can you compile in Debug mode and tell me at which line the segfault occurs?
Did you change any of the default paramters?

seboz123, 2019-11-12:
I changed the input topic to /kitti/velo/pointcloud to work with the kitti dataset.

`Segmenting cloud with 122550 points...
[New Thread 0x7fffd1a5a700 (LWP 6455)]
[Thread 0x7fffd1a5a700 (LWP 6455) exited]
[New Thread 0x7fffd1259700 (LWP 6456)]
[Thread 0x7fffd1259700 (LWP 6456) exited]
[New Thread 0x7fffd1259700 (LWP 6457)]
[New Thread 0x7fffd1a5a700 (LWP 6458)]
[Thread 0x7fffd1a5a700 (LWP 6458) exited]
[Thread 0x7fffd1259700 (LWP 6457) exited]
[New Thread 0x7fffd1a5a700 (LWP 6459)]
[New Thread 0x7fffd1259700 (LWP 6460)]
[Thread 0x7fffd1259700 (LWP 6460) exited]

Thread 10 "ground_segmenta" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fffd1a5a700 (LWP 6459)]
0x00007ffff685d16b in Segment::verticalDistanceToLine (this=0x555555852bc0, d=@0x7fffd1a597e8: 16.31100047735471, z=@0x7fffd1a597e0: -2.000999927520752)
    at /workspaces/ros_ws_melodic/src/packages/linefit_ground_segmentation/linefit_ground_segmentation/src/segment.cc:108
108         if (it->first.d - kMargin < d && it->second.d + kMargin > d) {
(gdb) [ground_segmentation-1] killing on exit
Quit`

That is the output of gdb when I run the code in the gdb debugger.

lorenwel, 2019-11-12:
I do not see any obvious reason how a segfault could occur there.
The line where the error occurs is line number 104 on the master branch. Did you make any changes to the code itself?

Could you also tell me the exact Kitti sequence you're using? Maybe I can reproduce the error myself.

seboz123, 2019-11-13:
No I just pulled the fresh repo again with just the input topic chaged to /kitti/velo/pointcloud 
The error still happens. I am using the kitti drive_0005 which I converted to a rosbag via the kitti2rosbag converter.
Your package also does not run on my own bagfiles unfortunately.
It seems that the number of threads until it crashes differs. Now Thread 27 crashes:
```
Thread 27 "ground_segmenta" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fffd1a5a700 (LWP 11604)]
0x00007ffff685d16b in Segment::verticalDistanceToLine (this=0x555555852bc0, d=@0x7fffd1a597e8: 16.31100047735471, z=@0x7fffd1a597e0: -2.000999927520752)
    at /workspaces/ros_ws_melodic/src/packages/linefit_ground_segmentation/linefit_ground_segmentation/src/segment.cc:104
104         if (it->first.d - kMargin < d && it->second.d + kMargin > d) {
(gdb) [ground_segmentation-1] killing on exit
Quit

```

lorenwel, 2019-11-13:
The `fix/out_of_bounds_segment` branch of PR #22 should fix this. Can you please verify, then I'll merge to master. 
[image]

seboz123, 2019-11-14:
This seems to fix it. Thank you very much!
Which parameters did you use for the posted gif?
Thanks in advance.

lorenwel, 2019-11-14:
I used the default parameters, except for `sensor_height: 1.8` and `r_max: 50.0`

Thanks for reporting the bug!

closed 2019-11-14 by lorenwel