lorenwel/linefit_ground_segmentation #29: Points obvious in the ground plane were treated as obstacle points, why?

bxc237, 2020-12-23:
Hi, I am using this segmentation package for point clouds capturing by the stereo camera and found that there are some speckles in the ground were treated as obstacle points. Those speckles were typically located on the edge of camera's FOV (please check the picture, in this scene, there is a little slope and some small items on it, orange points are obstacle_cloud), and my yaml look like this:
````
n_threads: 4            # number of threads to use.

r_min: 0.05              # minimum point distance. 0.5
r_max: 6              # maximum point distance. 50
n_bins: 120             # number of radial bins. 120
n_segments: 90         # number of radial segments. 360

max_dist_to_line: 0.03  # maximum vertical distance of point to line to be considered ground. 0.05

sensor_height: 0.075      # sensor height above ground. 1.8
max_slope: 0.3          # maximum slope of a ground line. 0.3
max_fit_error: 0.05     # maximum error of a point during line fit. 0.05
long_threshold: 0.1    # distance between points after which they are considered far from each other. 1.0
max_long_height: 0.1    # maximum height change to previous point in long line. 0.1
max_start_height: 0.2  # maximum difference to estimated ground height to start a new line. 0.2
line_search_angle: 0.1  # how far to search in angular direction to find a line [rad]. 0.1

latch: false            # latch output topics or not
visualize: false        # visualize segmentation result - USE ONLY FOR DEBUGGING
````
 I will be really appreciated if anyone can help me solve this issue. Thanks!  
[image]

lorenwel, 2021-02-02:
Hi, @bxc237 
this algorithm is designed for lidar, so I don't know what kind of performance to expect with depth/disparity images. 
You could try reducing `n_bins` (maybe to 60 or 30), given that you have a maximum distances of only 6 meters. With small bins noise in the point cloud can more easily lead to larger slopes of the fitted line.

closed 2021-02-02 by lorenwel

bxc237, 2021-03-04:
Hello, @lorenwel 
I have already reduced `n_bins`, but the performance is the same. I attach the PCL visualization result here. In the point clouds cluster that I highlighted (a piece of ground),  no ground lines can be generated, but still, some points (red points) are considered as ground points, some (white points) are not. Could you give me some suggestions about this issue?     
[image]
zoom in,
[image]