lorenwel/linefit_ground_segmentation #31: catkin_make fails

bruchpilot123, 2021-05-24:
Hello, 

during catkin_make i receive the following warnings:

__-- Could NOT find ensenso (missing: ENSENSO_LIBRARY ENSENSO_INCLUDE_DIR)
** WARNING ** io features related to ensenso will be disabled
-- Could NOT find DAVIDSDK (missing: DAVIDSDK_LIBRARY DAVIDSDK_INCLUDE_DIR)
** WARNING ** io features related to davidSDK will be disabled
-- Could NOT find DSSDK (missing: _DSSDK_LIBRARIES)
** WARNING ** io features related to dssdk will be disabled
** WARNING ** io features related to pcap will be disabled
** WARNING ** io features related to png will be disabled
-- The imported target "vtk" references the file
"/usr/bin/vtk"
but this file does not exist. Possible reasons include:

The file was deleted, renamed, or moved to another location.
An install or uninstall procedure did not complete successfully.
The installation package was faulty and contained
"/usr/lib/cmake/vtk-6.2/VTKTargets.cmake"
but not all the files it references.__

I could solve the ensenso and DAVIDSDK, with installing ENSENSO und DAVIDSDK. Finally catkin_make fail with:
CMake Error at as_ros/src/linefit_ground_segmentation/linefit_ground_segmentation_ros/CMakeLists.txt:29 (add_custom_target):
  add_custom_target cannot create target "headers" because another target
  with the same name already exists.  The existing target is a custom target
  created in source directory
  "/home/robert/catkin_ws/src/as_ros/src/linefit_ground_segmentation/linefit_ground_segmentation".
  See documentation for policy CMP0002 for more details.

I would really appreciate your help. I am using ROS Melodic and Ubuntu 18.04. All Prerequirements should be completed.

Kindly regards

lorenwel, 2021-05-27:
It appears as if it is pulling some mismatched dependency. 
Could you please try building with python catkin tools?
Install them with `sudo apt install python-catkin-tools` and then build with `catkin build`. I do not use `catkin_make` myself and therefore didn't test it. 
Make sure to clean your workspace before building with `catkin build`.

lorenwel, 2021-05-27:
I did some digging myself and appears as if this issue comes from the pcl::visualizations dependency. It seems to be sensitive to the VTK version which you have installed. [See here](https://github.com/PointCloudLibrary/pcl/issues/3525).

bruchpilot123, 2021-05-29:
Good evening, 
i just built the workspace with catkin build and it also did not work:

h:6:10: fatal error: glog/logging.h: No such file or directory
 #include <glog/logging.h>
Actually i do not know why this happens, because catkin build says that it was able to Start and Finish building glog_catkin.
I am just gonna check, the other comment tomorrow with the vtk version. 

Kindy regards

lorenwel, 2021-06-11:
@bruchpilot123 Do you have any update on this?

bruchpilot123, 2021-06-11:
I currently have exams, i made some progress but i will follow with a detailed update. Please do not close the issue.

Kindly regards

sarahnator, 2021-07-14:
Hi @lorenwel, I have the same setup as @bruchpilot123 (melodic, ubuntu 18.04) and am also experiencing the same issue with  glog/logging.h  not being found after running `catkin build`. There are no issues being raised about the vtk that I can see. The readme doesn't mention if compiling the dependencies is required -- if that's needed, it is possible I did it incorrectly.

Do either of you have suggestions or workarounds?

lorenwel, 2021-07-23:
Hi both, 

the glog dependency is actually obsolete and has been for some while. I finally removed it in a new feature branch #32. Could you please check whether this also fixes your issues?

lorenwel, 2022-03-10:
I'll assume that this issue has been fixed when merging #32. Please reopen if this issue persits.

closed 2022-03-10 by lorenwel