lorenwel/linefit_ground_segmentation #36: Output stuck on first frame

Mnzs1701, 2022-04-18:
Thanks for your excellent work.
I was trying to implement this code however the output is always stuck on the first frame (when using the visualizer) and there is no points on the output topic when viewed in RViz. This happens with both the provided file and with my own dataset.
The points appear to be segmented but it is only the first frame that is visible. 
The output on screen simply says 

> Segmenting cloud with 122527 points...

I used catkin build configured to 'Release' type. The 3D Viewer indicates around 70 FPS however the output is frozen

lorenwel, 2022-04-19:
Hi,
the `visualize` flag is meant only for debugging and will therefore only show a single point cloud. 
If you want to continuously visualize point clouds you should set `visualize` to False and use Rviz.

Mnzs1701, 2022-04-20:
Thanks for the clarification. I had initially copied only the `linefit_ground_segmentation` and `linefit_ground_segmentation_ros` folders into another workspace, this might have caused some problem
I rebuilt in a new workspace after cloning the entire package along with `catkin_simple` package. Although `catkin_make` fails `catkin build` works perfectly.
The issue is resolved and I am getting very clear ground plane segmentation on an Ouster OS1-128 Lidar with an average time of 10 ms. 
Thanks for all the great work and help.

closed 2022-04-20 by Mnzs1701

lorenwel, 2022-04-21:
Good to hear. FYI, I improved the visualizer to update live in the meantime (see #37)