lorenwel/linefit_ground_segmentation #39: Which version of pcl is recommended?

takeyamayuki, 2022-07-22:
Hello.
Thank you for your great project.

I got the following error when I `catkin build` in my environment (WSL2, Ubuntu20.04, ROS noetic).
The file structure is as follows.
```
catkin_ws
└── src
    ├── linefit_ground_segmentation
    │   ├── CMakeLists.txt
    │   ├── include
    │   │   └── ground_segmentation
    │   │       ├── bin.h
    │   │       ├── ground_segmentation.h
    │   │       ├── segment.h
    │   │       ├── typedefs.h
    │   │       └── viewer.h
    │   ├── package.xml
    │   └── src
    │       ├── bin.cc
    │       ├── ground_segmentation.cc
    │       ├── segment.cc
    │       └── viewer.cc
    ├── linefit_ground_segmentation_ros
    │   ├── CMakeLists.txt
    │   ├── launch
    │   │   ├── segmentation.launch
    │   │   ├── segmentation_params.yaml
    │   │   └── test.launch
    │   ├── package.xml
    │   └── src
    │       ├── ground_segmentation_node.cc
    │       └── ground_segmentation_test_node.cc
```

```sh
_________________________________________________________________________________________________________________________________________
Errors     << linefit_ground_segmentation:make /home/yuki/catkin_ws/logs/linefit_ground_segmentation/build.make.000.log
/home/yuki/catkin_ws/src/linefit_ground_segmentation/src/ground_segmentation.cc: In member function ‘void GroundSegmentation::segment(const PointCloud&, std::vector<int>*)’:
/home/yuki/catkin_ws/src/linefit_ground_segmentation/src/ground_segmentation.cc:51:66: error: conversion from ‘std::shared_ptr<pcl::PointCloud<pcl::PointXYZ> >’ to non-scalar type ‘pcl::PointCloud<pcl::PointXYZ>::Ptr’ {aka ‘boost::shared_ptr<pcl::PointCloud<pcl::PointXYZ> >’} requested
   51 |     PointCloud::Ptr obstacle_cloud = std::make_shared<PointCloud>();
      |                                      ~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~
/home/yuki/catkin_ws/src/linefit_ground_segmentation/src/ground_segmentation.cc:54:64: error: conversion from ‘std::shared_ptr<pcl::PointCloud<pcl::PointXYZ> >’ to non-scalar type ‘pcl::PointCloud<pcl::PointXYZ>::Ptr’ {aka ‘boost::shared_ptr<pcl::PointCloud<pcl::PointXYZ> >’} requested
   54 |     PointCloud::Ptr ground_cloud = std::make_shared<PointCloud>();
      |                                    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~
/home/yuki/catkin_ws/src/linefit_ground_segmentation/src/ground_segmentation.cc:60:61: error: conversion from ‘std::shared_ptr<pcl::PointCloud<pcl::PointXYZ> >’ to non-scalar type ‘pcl::PointCloud<pcl::PointXYZ>::Ptr’ {aka ‘boost::shared_ptr<pcl::PointCloud<pcl::PointXYZ> >’} requested
   60 |     PointCloud::Ptr min_cloud = std::make_shared<PointCloud>();
      |                                 ~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~
make[2]: *** [CMakeFiles/linefit_ground_segmentation.dir/build.make:63: CMakeFiles/linefit_ground_segmentation.dir/src/ground_segmentation.cc.o] エラー 1
make[2]: *** 未完了のジョブを待っています....
make[1]: *** [CMakeFiles/Makefile2:305: CMakeFiles/linefit_ground_segmentation.dir/all] エラー 2
make: *** [Makefile:141: all] エラー 2
cd /home/yuki/catkin_ws/build/linefit_ground_segmentation; catkin build --get-env linefit_ground_segmentation | catkin env -si  /usr/bin/make --jobserver-auth=3,4; cd -

.........................................................................................................................................
Failed     << linefit_ground_segmentation:make               [ Exited with code 2 ]
Failed    <<< linefit_ground_segmentation                    [ 14.4 seconds ]
Abandoned <<< linefit_ground_segmentation_ros                [ Unrelated job failed ]
```

This is probably a problem because the desired pcl version is not installed.
And the pcl version in my environment is described as `Version: 1.10.0+dfsg-5ubuntu1` when checked with `apt show libpcl-dev`.

Which version is the author using?
If it is not a version difference, what is the problem?

Thank you in advance.

lorenwel, 2022-07-22:
Hi @takeyamayuki,

thanks for opening the issue and describing the issue well. PCL appears to have [introduced its own shared_ptr definition](https://github.com/PointCloudLibrary/pcl/blob/pcl-1.10.0/common/include/pcl/pcl_macros.h#L90) with version 1.10, which causes this issue.
Could you please check if #40 fixes this for you? That worked for me.

takeyamayuki, 2022-07-22:
Thanks for the quick response.

I tried #40 and got the following error and notes.
(An error is only at the top)

```sh
_________________________________________________________________________________________________________________________________________
Errors     << linefit_ground_segmentation:make /home/yuki/catkin_ws/logs/linefit_ground_segmentation/build.make.002.log
/home/yuki/catkin_ws/src/linefit_ground_segmentation/src/viewer.cc: In member function ‘void Viewer::visualizePointCloud(const PointCloudConstPtr&, const string&)’:
/home/yuki/catkin_ws/src/linefit_ground_segmentation/src/viewer.cc:76:37: error: no matching function for call to ‘pcl::visualization::PCLVisualizer::updatePointCloud(const PointCloudConstPtr&, const string&)’
   76 |   viewer_.updatePointCloud(cloud, id);
      |                                     ^
In file included from /home/yuki/catkin_ws/src/linefit_ground_segmentation/include/ground_segmentation/viewer.h:8,
                 from /home/yuki/catkin_ws/src/linefit_ground_segmentation/src/viewer.cc:1:
/usr/include/pcl-1.10/pcl/visualization/pcl_visualizer.h:680:9: note: candidate: ‘template<class PointT> bool pcl::visualization::PCLVisualizer::updatePointCloud(const typename pcl::PointCloud<PointT>::ConstPtr&, const string&)’
  680 |         updatePointCloud (const typename pcl::PointCloud<PointT>::ConstPtr &cloud,
      |         ^~~~~~~~~~~~~~~~
/usr/include/pcl-1.10/pcl/visualization/pcl_visualizer.h:680:9: note:   template argument deduction/substitution failed:
/home/yuki/catkin_ws/src/linefit_ground_segmentation/src/viewer.cc:76:37: note:   couldn’t deduce template parameter ‘PointT’
   76 |   viewer_.updatePointCloud(cloud, id);
      |                                     ^
In file included from /home/yuki/catkin_ws/src/linefit_ground_segmentation/include/ground_segmentation/viewer.h:8,
                 from /home/yuki/catkin_ws/src/linefit_ground_segmentation/src/viewer.cc:1:
/usr/include/pcl-1.10/pcl/visualization/pcl_visualizer.h:690:9: note: candidate: ‘template<class PointT> bool pcl::visualization::PCLVisualizer::updatePointCloud(const typename pcl::PointCloud<PointT>::ConstPtr&, const pcl::visualization::PointCloudGeometryHandler<PointT>&, const string&)’
  690 |         updatePointCloud (const typename pcl::PointCloud<PointT>::ConstPtr &cloud,
      |         ^~~~~~~~~~~~~~~~
/usr/include/pcl-1.10/pcl/visualization/pcl_visualizer.h:690:9: note:   template argument deduction/substitution failed:
/home/yuki/catkin_ws/src/linefit_ground_segmentation/src/viewer.cc:76:37: note:   ‘const string’ {aka ‘const std::__cxx11::basic_string<char>’} is not derived from ‘const pcl::visualization::PointCloudGeometryHandler<PointT>’
   76 |   viewer_.updatePointCloud(cloud, id);
      |                                     ^
In file included from /home/yuki/catkin_ws/src/linefit_ground_segmentation/include/ground_segmentation/viewer.h:8,
                 from /home/yuki/catkin_ws/src/linefit_ground_segmentation/src/viewer.cc:1:
/usr/include/pcl-1.10/pcl/visualization/pcl_visualizer.h:701:9: note: candidate: ‘template<class PointT> bool pcl::visualization::PCLVisualizer::updatePointCloud(const typename pcl::PointCloud<PointT>::ConstPtr&, const pcl::visualization::PointCloudColorHandler<PointT>&, const string&)’
  701 |         updatePointCloud (const typename pcl::PointCloud<PointT>::ConstPtr &cloud,
      |         ^~~~~~~~~~~~~~~~
/usr/include/pcl-1.10/pcl/visualization/pcl_visualizer.h:701:9: note:   template argument deduction/substitution failed:
/home/yuki/catkin_ws/src/linefit_ground_segmentation/src/viewer.cc:76:37: note:   ‘const string’ {aka ‘const std::__cxx11::basic_string<char>’} is not derived from ‘const pcl::visualization::PointCloudColorHandler<PointT>’
   76 |   viewer_.updatePointCloud(cloud, id);
      |                                     ^
In file included from /home/yuki/catkin_ws/src/linefit_ground_segmentation/include/ground_segmentation/viewer.h:8,
                 from /home/yuki/catkin_ws/src/linefit_ground_segmentation/src/viewer.cc:1:
/usr/include/pcl-1.10/pcl/visualization/pcl_visualizer.h:916:9: note: candidate: ‘bool pcl::visualization::PCLVisualizer::updatePointCloud(const ConstPtr&, const string&)’
  916 |         updatePointCloud (const pcl::PointCloud<pcl::PointXYZ>::ConstPtr &cloud,
      |         ^~~~~~~~~~~~~~~~
/usr/include/pcl-1.10/pcl/visualization/pcl_visualizer.h:916:75: note:   no known conversion for argument 1 from ‘const PointCloudConstPt
’ {aka ‘const std::shared_ptr<const pcl::PointCloud<pcl::PointXYZ> >’} to ‘const ConstPtr&’ {aka ‘const boost::shared_ptr<const pcl::PointCloud<pcl::PointXYZ> >&’}
  916 |         updatePointCloud (const pcl::PointCloud<pcl::PointXYZ>::ConstPtr &cloud,
      |                           ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~
/usr/include/pcl-1.10/pcl/visualization/pcl_visualizer.h:928:9: note: candidate: ‘bool pcl::visualization::PCLVisualizer::updatePointCloud(const ConstPtr&, const string&)’
  928 |         updatePointCloud (const pcl::PointCloud<pcl::PointXYZRGB>::ConstPtr &cloud,
      |         ^~~~~~~~~~~~~~~~
/usr/include/pcl-1.10/pcl/visualization/pcl_visualizer.h:928:78: note:   no known conversion for argument 1 from ‘const PointCloudConstPt
’ {aka ‘const std::shared_ptr<const pcl::PointCloud<pcl::PointXYZ> >’} to ‘const ConstPtr&’ {aka ‘const boost::shared_ptr<const pcl::PointCloud<pcl::PointXYZRGB> >&’}
[... cut ...]
.........................................................................................................................................
Failed     << linefit_ground_segmentation:make               [ Exited with code 2 ]
Failed    <<< linefit_ground_segmentation                    [ 4.9 seconds ]
Abandoned <<< linefit_ground_segmentation_ros                [ Unrelated job failed ]
```

After this, I tried to tweak the code in various ways and managed to `catkin build` in my environment, so I would like to submit a Pull Request once I have it.

Thank you in advance.

takeyamayuki, 2022-07-22:
I'm not very familiar with github issues and pull rquests, so I'm sorry if there is something wrong.

lorenwel, 2022-07-26:
I made a change to #40. Now it should finally, actually work.

PR #40 (2022-07-27): Fixed shared_ptr issue introduced with libpcl-1.10
Fixes #39
closed 2022-07-27 by lorenwel