lorenwel/linefit_ground_segmentation #42: Issue whan launching segmentation.node

aldipiroli, 2022-08-15:
Hello and thanks for the great work!
I wanted to test the repo but I cannot somehow launch the node because I get this error:

`roslaunch linefit_ground_segmentation_ros segmentation.launch`

```
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://localhost:42143/

SUMMARY
========

PARAMETERS
 * /ground_segmentation/gravity_aligned_frame: 
 * /ground_segmentation/ground_output_topic: ground_cloud
 * /ground_segmentation/input_topic: /kitti/velo/point...
 * /ground_segmentation/latch: False
 * /ground_segmentation/line_search_angle: 0.1
 * /ground_segmentation/long_threshold: 1.0
 * /ground_segmentation/max_dist_to_line: 0.05
 * /ground_segmentation/max_fit_error: 0.05
 * /ground_segmentation/max_long_height: 0.1
 * /ground_segmentation/max_slope: 0.3
 * /ground_segmentation/max_start_height: 0.2
 * /ground_segmentation/min_slope: 0.0
 * /ground_segmentation/n_bins: 120
 * /ground_segmentation/n_segments: 360
 * /ground_segmentation/n_threads: 4
 * /ground_segmentation/obstacle_output_topic: obstacle_cloud
 * /ground_segmentation/r_max: 50
 * /ground_segmentation/r_min: 0.5
 * /ground_segmentation/sensor_height: 1.8
 * /ground_segmentation/visualize: False
 * /rosdistro: noetic
 * /rosversion: 1.15.14

NODES
  /
    ground_segmentation (linefit_ground_segmentation_ros/ground_segmentation_node)

auto-starting new master
process[master]: started with pid [17201]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 70bd43ea-1cac-11ed-8083-04d9f5f773c7
process[rosout-1]: started with pid [17226]
started core service [/rosout]
process[ground_segmentation-2]: started with pid [17233]
/home/catkin_ws/devel/lib/linefit_ground_segmentation_ros/ground_segmentation_node: symbol lookup error: /home/catkin_ws/devel/lib/liblinefit_ground_segmentation.so: undefined symbol: _ZNK3pcl13visualization28PointCloudGeometryHandlerXYZINS_8PointXYZEE11getGeometryER15vtkSmartPointerI9vtkPointsE
[ground_segmentation-2] process has died 
```

Not sure if it has anything to do with the repo, this were my commands:

```
sudo apt install ros-noetic-eigen-conversions
mkdir -p catkin_ws/src
cd catkin_ws && catkin init
cd src
git clone https://github.com/catkin/catkin_simple.git
git clone https://github.com/lorenwel/linefit_ground_segmentation.git
cd ..
catkin build 
source devel/setup.bash
roslaunch linefit_ground_segmentation_ros segmentation.launch
```

lorenwel, 2022-08-25:
@aldipiroli Did your issue resolve?

lorenwel, 2022-08-29:
Please reopen if it is still an issue.

closed 2022-08-29 by lorenwel