lorenwel/linefit_ground_segmentation #47: building

jacobsayono, 2023-02-28:
hi, can you please walk me through the steps on how to build this?
i'm trying to use this without ROS.
do i put catkin_simple into linefit_ground_segmentation/linefit_ground_segmentation?
what do i do next?

closed 2023-02-28 by lorenwel

jacobsayono, 2023-03-05:
hi, thanks for your response. i am trying to do this **without** ROS. i assume that i will be using the folder `linefit_ground_segmentation` instead of `linefit_ground_segmentation_ros`. in that case, would i still need to use `catkin_simple` to build?

i am working on a research project, which does not involve ROS. i am merely trying to use a quick segmentation algorithm to separate the ground point clouds from all the others -- this is to develop a traversibility algorithm later on.

is this possible with just pure c++ and no ROS? thanks and i hope i can use this for my project so that i can cite the paper!

lorenwel, 2023-03-06:
At the moment, even without ROS, you will still need the [catkin build system to build this package](https://catkin-tools.readthedocs.io/en/latest/installing.html). But I wanted to remove the dependency on catkin_simple for a while now, maybe I'll take this as a push to finally do that. Then it should be possible to build the library with just regular CMake.

PS: Not my paper, so a prospective citation is not an incentive for me ;)

jacobsayono, 2023-03-07:
i see thank you for your help. i will definitely reference this repository to my work then!

i'm having troubles with `catkin_simple` because it's not quite documented and i am stuck with the entire build process for this repository.
would you be able to walk through the steps with me?

here's the workflow i did so far:
```
git clone https://github.com/lorenwel/linefit_ground_segmentation.git
cd linefit_ground_segmentation/linefit_ground_segmentation
cd src
git clone https://github.com/catkin/catkin_simple.git
cd ..
catkin build
```

lorenwel, 2023-03-07:
Hi, I removed `catkin_simple` in #49. Maybe that helps you. 
I checked that it build like this
```
cd linefit_ground_segmentation
mkdir build 
cd build
cmake .. 
make
```
I did not check if I can successfully build a pure CMake function against it, you will have to try yourself. 

If you want to try building it with `catkin` please refer to the introduction video I linked in my first response.

jacobsayono, 2023-03-08:
thank you so much for doing this!

i think you committed but did not accept your own pull request. i tried to clone your remove_catkin_simple branch but it's not found in upstream origin.

i will try to use noetic to launch this ros package later and let you know my progress.

i would also like to ask how i can test a point-cloud file? your ros version allows you to test a `kitti.pyl` file in /docs
i'm not sure how i can test an input file with the pure c++ version

also, can it take in .pcd files by any chance? or only .pyl?
there's probably a converter somewhere out there, but just wondering!