lorenwel/linefit_ground_segmentation #48: somthing wrong when using [Viewer] in test file

Erisnoit, 2023-03-06:
hey, thanks for your great work.  but there're sth wrong when using it, and i still have some problems after debugging.

1.run the test_node.launch, crash before seg(almost after the visual window flash), 
and get the bug as follows(debug output using GBD)
p.s.nothing changed in this work.
[image]
so it seems that sth wrong about vtk/visualize

2.close the visualize func, and it runs well(can get point segmentation)
so the problem seems in viewer class .
cause it crash before seg and visual func in segmentation file, the problem may be in this part?(please correct me if wrong)
[image]
i don't really konw what this part means, especially why 2 for-loop??
could you please explain this part?

3.my pcl is 1.8 and vtk is 6.3, what's the recommand version?
 is this problem caused by pkg version?

hope you can give me some advice. thanks a lot!

lorenwel, 2023-03-06:
Hi @Erisnoit 
given that the error occurs inside of the visualizer, I suspect that the issue is related to your PCL installation and not necessarily an incompatibility with this code. 
Can you please try to run the PCL visualizer demo, [as suggested here](https://github.com/lorenwel/linefit_ground_segmentation/issues/44#issuecomment-1238178983).
If that already fails it is not an issue with the code in this repo. 

Please also make sure that you really only have one instance of PCL installed/compiled on your machine. This frequently happens, since there are multiple ways to obtain PCL.

Erisnoit, 2023-03-07:
hi，thanks for your quick reply.
i test the PCLVisualizer demo as you mentioned, and it all works well in my PCL1.8 and its dependence vtk6.3.
[image]
[image]

indeed, there're 2 instance in my machine(PCL1.8 and PCL1.12), but i don't really think it counts, for i have been using this 2 instance for along time and they both run well in other works. 
besides, i set the PCL version to1.8 in both your work and PCLVisualizer demo.
[image]
is there any other method to find the problem?
THANKS.

Erisnoit, 2023-03-07:
additionally, i just closed the viewer thread and changed it directly using spinOnce(), and it worked
[image]
the changes are as follows, in Viewer constructor and in Viewer::visualize() func
[image]
[image]

but i still think it a nice idea to make a standalone viewer thread, 
could you help me figure out sth wrong  with this thread?or with my configuration??

lorenwel, 2023-03-07:
Ok, in that case it seems to be an issue with my code. The reason for having a separate viewer thread is that we can visualize things in real-time, for example when playing back a ROS bag. That was introduced in #37 before that the viewer was also blocking, as it is in your code snippet.

Can you please run a backtrace on the segfault you encounter? That will give me a bit more information. 
Did you modify the code at all, before encountering the issue?
Do you just run `roslaunch linefit_ground_segmentation_ros test.launch` ?
Did you modify any parameters?

Erisnoit, 2023-03-07:
i use gbd backtrace, and here is some information.
[image]
[image]

it seems that #16 is located in your code.
i just add cout statement(use to locate), no other modification.
just run test.launch, and just use demo data.
no params modification.
thanks!

lorenwel, 2023-03-07:
I am able to reproduce a segfault, but it it happens only in Debug mode and seems unrelated. 
Did you build in Release or Debug mode before? If you built in Debug (or didn't specify), please try building in Release:
```
catkin config --cmake-args -DCMAKE_BUILD_TYPE=RelWithDebInfo
catkin build
```

Erisnoit, 2023-03-08:
i directly used catkin_make and SET(CMAKE_BUILD_TYPE "Release") in all CMakeList.txt before,
and the segfault happened in both type.

then, i also followed your advice and tried it agin, use catkin build and set the config,
(i'm sure it's in Release type, from time cost output)
but nothing changed when i turn on the visualize, segfault is still there...